adafruit/Adafruit_nRF52_Arduino #722: Compile errors with seesaw library

rdmcmurray, 2022-03-16:
### Operating System

MacOS

### IDE version

1.8.19

### Board

Feather nRF52840 Express

### BSP version

S140 6.1.1

### Sketch

`#include <Adafruit_seesaw.h>

void setup() 
{
  //
}

void loop() 
{
  //
}`

### What happened ?

Sketch fails to compile when the seesaw library is present on the nRF52840 in Arduino.

### How to reproduce ?

1. Include seesaw v1.6.0 via Library Manager in a blank Arduino IDE sketch
2. Set board to nRF52840 Express
3. Verify

### Debug Log

In file included from /Users/rmc/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp:1:
/Users/rmc/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.h:33:27: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope; did you mean 'NRF_SPIM_BIT_ORDER_MSB_FIRST'?
   33 |        BitOrder dataOrder=SPI_BITORDER_MSBFIRST,
      |                           ^~~~~~~~~~~~~~~~~~~~~
      |                           NRF_SPIM_BIT_ORDER_MSB_FIRST
In file included from /Users/rmc/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_BusIO_Register.h:2,
                 from /Users/rmc/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_BusIO_Register.cpp:1:
/Users/rmc/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.h:33:27: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope; did you mean 'NRF_SPIM_BIT_ORDER_MSB_FIRST'?
   33 |        BitOrder dataOrder=SPI_BITORDER_MSBFIRST,
      |                           ^~~~~~~~~~~~~~~~~~~~~
      |                           NRF_SPIM_BIT_ORDER_MSB_FIRST
/Users/rmc/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.h:39:27: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope; did you mean 'NRF_SPIM_BIT_ORDER_MSB_FIRST'?
   39 |        BitOrder dataOrder=SPI_BITORDER_MSBFIRST,
      |                           ^~~~~~~~~~~~~~~~~~~~~
      |                           NRF_SPIM_BIT_ORDER_MSB_FIRST
/Users/rmc/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.h:39:27: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope; did you mean 'NRF_SPIM_BIT_ORDER_MSB_FIRST'?
   39 |        BitOrder dataOrder=SPI_BITORDER_MSBFIRST,
      |                           ^~~~~~~~~~~~~~~~~~~~~
      |                           NRF_SPIM_BIT_ORDER_MSB_FIRST
/Users/rmc/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp: In member function 'void Adafruit_SPIDevice::transfer(uint8_t*, size_t)':
/Users/rmc/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp:100:23: error: 'SPI_BITORDER_LSBFIRST' was not declared in this scope; did you mean 'NRF_SPIM_BIT_ORDER_LSB_FIRST'?
  100 |     if (_dataOrder == SPI_BITORDER_LSBFIRST) {
      |                       ^~~~~~~~~~~~~~~~~~~~~
      |                       NRF_SPIM_BIT_ORDER_LSB_FIRST
/Users/rmc/Documents/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp:128:23: error: 'SPI_BITORDER_LSBFIRST' was not declared in this scope; did you mean 'NRF_SPIM_BIT_ORDER_LSB_FIRST'?
  128 |     if (_dataOrder == SPI_BITORDER_LSBFIRST) {
      |                       ^~~~~~~~~~~~~~~~~~~~~
      |                       NRF_SPIM_BIT_ORDER_LSB_FIRST
exit status 1
Error compiling for board Adafruit Feather nRF52840 Express.

hathach, 2022-03-16:
just tested, and couldn't reproduce the issue, 
1. which nrf52 BSP version you are on ? the version displayed in the "Board Manager" not the SD version. 
2. attached txt with your full compiled log

ladyada, 2022-03-16:
please check you updated all libraries in library manger :)

rdmcmurray, 2022-03-16:
nrf52 BSP is 1.3.0. Everything is fresh and no boards or libraries are un-updated. I'll nuke it and start over.

hathach, 2022-03-16:
don't forget to attach your compile log

closed 2022-03-16 by rdmcmurray