adafruit/Adafruit_nRF52_Arduino #723: Unable to use gamepad features for Adafruit Feather nrf52832

4a454646, 2022-03-28:
### Operating System

Linux

### IDE version

1.18.19

### Board

Adafruit Feather nrf52832

### BSP version

Github latest

### Sketch

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/Peripheral/blehid_gamepad/blehid_gamepad.ino

### What happened ?

I am able to successfully connect over Bluetooth to my board, however, it is classified as type "Unknown" and none of the gamepad features, like the joystick or buttons, register. The same code works fine on Windows. 

### How to reproduce ?

1. Upload basic starter code to board
2. Connect board via bluetooth to computer
3. The device is initially recognized as a gamepad, but changes to Unknown, and none of the inputs work.

### Debug Log

I suspect it is something to do with a faulty device descriptor. See below:
```
jeff@pop-os:~$ sudo systemctl status bluetooth
● bluetooth.service - Bluetooth service
     Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2022-03-27 19:35:03 PDT; 1min 45s ago
       Docs: man:bluetoothd(8)
   Main PID: 8024 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 16507)
     Memory: 1.2M
     CGroup: /system.slice/bluetooth.service
             └─8024 /usr/lib/bluetooth/bluetoothd

Mar 27 19:36:34 pop-os bluetoothd[8024]: profiles/input/hog-lib.c:report_map_read_cb() Report Map read failed: Invalid handle
Mar 27 19:36:34 pop-os bluetoothd[8024]: profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Invalid handle
Mar 27 19:36:35 pop-os bluetoothd[8024]: profiles/input/hog-lib.c:report_reference_cb() Malformed ATT read response
Mar 27 19:36:41 pop-os bluetoothd[8024]: profiles/input/hog-lib.c:proto_mode_read_cb() ATT protocol error
Mar 27 19:36:41 pop-os bluetoothd[8024]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Attribute can't be read
Mar 27 19:36:41 pop-os bluetoothd[8024]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Invalid handle
Mar 27 19:36:41 pop-os bluetoothd[8024]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Invalid handle
Mar 27 19:36:41 pop-os bluetoothd[8024]: profiles/input/hog-lib.c:report_map_read_cb() Report Map read failed: Invalid handle
Mar 27 19:36:41 pop-os bluetoothd[8024]: profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Invalid handle
Mar 27 19:36:41 pop-os bluetoothd[8024]: profiles/input/hog-lib.c:report_reference_cb() Malformed ATT read response
```

### Screenshots

https://i.imgur.com/XnU79nS.png
https://i.imgur.com/TYw3PZ1.png
https://i.imgur.com/GXWnf3O.png

hathach, 2022-03-28:
can you post the log from the nrf52 side with debug level = 2

4a454646, 2022-03-28:
```
Bluefruit52 HID Gamepad Example
-------------------------------

Go to your phone's Bluetooth settings to pair your device
then open an application that accepts gamepad input
[CFG   ] SoftDevice's RAM requires: 0x20002930

BSP Library : 1.3.0
Bootloader  : s132 6.1.1
Serial No   : 2744640E805EAE3A

--------- SoftDevice Config ---------
Max UUID128     : 10
ATTR Table Size : 3072
Service Changed : 1
Peripheral Connect Setting
  - Max MTU         : 23
  - Event Length    : 3
  - HVN Queue Size  : 1
  - WrCmd Queue Size: 1

--------- BLE Settings ---------
Name            : Feather nRF52832
Max Connections : Peripheral = 1, Central = 0 
Address         : D1:EA:17:10:68:36 (Static)
TX Power        : 4 dBm
Conn Intervals  : min = 11.25 ms, max = 15.00 ms
Conn Timeout    : 2000 ms
Peripheral Paired Devices: 
  3A5B37F60AE0 : pop-os (89 bytes)

[BLE   ] BLE_GAP_EVT_CONNECTED : Conn Handle = 0
[GAP   ] MAC = 64:B8:3D:6B:3F:38, Type = 2, Resolved = 0
[GAP   ] Conn Interval = 11.25 ms, Latency = 0, Supervisor Timeout = 420 ms
[BLE   ] BLE_GAP_EVT_SEC_PARAMS_REQUEST : Conn Handle = 0
[PAIR  ] Peer Params: bond = 1, mitm = 1, lesc = 1, io_caps = 1
[BLE   ] BLE_GATTS_EVT_EXCHANGE_MTU_REQUEST : Conn Handle = 0
[GAP   ] ATT MTU is changed to 23
[BLE   ] BLE_GAP_EVT_DATA_LENGTH_UPDATE_REQUEST : Conn Handle = 0
[GAP   ] Data Length Request is (tx, rx) octets = (251, 251), (tx, rx) time = (2120, 2120) us
[BLE   ] BLE_GAP_EVT_DATA_LENGTH_UPDATE : Conn Handle = 0
[GAP   ] Data Length is (tx, rx) octets = (27, 27), (tx, rx) time = (1364, 1364) us
[BLE   ] BLE_GAP_EVT_CONN_SEC_UPDATE : Conn Handle = 0
[BOND  ] CCCD setting not found
[PAIR  ] Security Mode = 1, Level = 2
[BLE   ] BLE_GAP_EVT_AUTH_STATUS : Conn Handle = 0
[PAIR  ] Auth Status = 0x00, Bonded = 1, LESC = 0, Our Kdist = 0x03, Peer Kdist = 0x03 
[BLE   ] BLE_GATTS_EVT_WRITE : Conn Handle = 0
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[BLE   ] BLE_GATTC_EVT_CHAR_VAL_BY_UUID_READ_RSP : Conn Handle = 0
[BOND  ] Saved keys for "pop-os" to file /adafruit/bond_prph/3A5B37F60AE0 ( 89 bytes )
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS

```

hathach, 2022-03-28:
I don't see any issue on device side, could do you detail on your linux machine (distro and version) and how do you test the gamepad. Last time I tried, it works just fine with ubuntu 20.04 using an app called 'antmicro'

4a454646, 2022-03-28:
```
jeff@pop-os:~$ neofetch
             /////////////                jeff@pop-os 
         /////////////////////            ----------- 
      ///////*767////////////////         OS: Pop!_OS 21.10 x86_64 
    //////7676767676*//////////////       Host: 82JY Legion 5 17ACH6H 
   /////76767//7676767//////////////      Kernel: 5.16.11-76051611-generic 
  /////767676///*76767///////////////     Uptime: 11 hours, 2 mins 
 ///////767676///76767.///7676*///////    Packages: 2659 (dpkg), 5 (flatpak) 
/////////767676//76767///767676////////   Shell: bash 5.1.8 
//////////76767676767////76767/////////   Resolution: 1920x1080 
///////////76767676//////7676//////////   DE: GNOME 40.5 
////////////,7676,///////767///////////   WM: Mutter 
/////////////*7676///////76////////////   WM Theme: Pop 
///////////////7676////////////////////   Theme: Pop-dark [GTK2/3] 
 ///////////////7676///767////////////    Icons: Uniform+ lightslategray folder 
  //////////////////////'////////////     Terminal: gnome-terminal 
   //////.7676767676767676767,//////      CPU: AMD Ryzen 7 5800H with Radeon Gr 
    /////767676767676767676767/////       GPU: AMD ATI 05:00.0 Cezanne 
      ///////////////////////////         Memory: 4970MiB / 13901MiB 
         /////////////////////
             /////////////                                        
                                                                 
```

Testing was done with an application called jstest-gtk, which recognizes my Xbox and Playstation controllers without issue but fails to see my custom gamepad. It is the same story with a website called [gamepad-tester](https://gamepad-tester.com/). I also tested it on one of my Steam games ([Crawl](https://store.steampowered.com/app/293780/Crawl/)) and it doesn't work there either. I can't seem to find the antmicro you're talking about, could you provide a link?

hathach, 2022-03-29:
[image]

just tested again, and the gamepad able to send buttons just fine with antimicro (mispelled in previous post). In screenshot, it is sending button5, antimicro is small gamepad test I found in ubuntu software store for testing

[image]

4a454646, 2022-04-01:
I actually have a later version of bluez, see here: 
```
jeff@pop-os:~$ apt list bluez
Listing... Done
bluez/impish-security,impish-updates,now 5.60-0ubuntu2.2 amd64 [installed]
bluez/impish-security,impish-updates 5.60-0ubuntu2.2 i386
```
Can you check that it still works on bluez 5.60? Bluez is tied to the Pop-desktop so I can't easily downgrade it, and I'd prefer not to mess with my kernel since doing so in the past has broken things like my wifi and graphics. Please check on your end that the gamepad still properly registers with bluez=5.60

hathach, 2022-04-02:
Unfortunately, I wouldn't want to upgrade my bluez stack either, especially the next LTS is about to dropped (I may go with Fedora 36 though). I will test again with the new OS later on. Meanwhile, maybe you could try to install 'AntiMicro' to test.