adafruit/Adafruit_nRF52_Arduino #725: WebUSB Serial is not working in nRF52840

milador, 2022-04-08:
### Operating System

Windows 10

### IDE version

1.8.19

### Board

Feather nRF52840 Sense

### BSP version

1.3.0

### Sketch

Examples/Adafruit_TinyUSB_Arduino/WebUSB/webusb-serial

### What happened ?

Doesn't connect to Feather Board and returns `NetworkError: Unable to claim interface.`

Tested in Chrome version 100

### How to reproduce ?

1. Go to File -> Examples -> Adafruit_TinyUSB_Arduino -> WebUSB -> webusb-serial
2. Compile and upload webusb-serial.ino to Feather nRF52840 Sense
3. Open `https://adafruit.github.io/Adafruit_TinyUSB_Arduino/examples/webusb-serial/index.html` in Chrome
4. Click on Connect

hathach, 2022-04-10:
- Have you tried to enable the webusb feature in chrome. I don’t remember the exact term, it is probably called **experimental web feature** or something. 
- Also the example did wait until the serial is opened, make sure you open the Serial monitor as well

https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/9527382b498cd1264d9132f9e7855434c869f557/examples/WebUSB/webusb_serial/webusb_serial.ino#L57

milador, 2022-04-11:
@hathach It was a chrome issue as it's working now. It works using Microsoft Edge as well.

closed 2022-04-11 by milador

hathach, 2022-04-11:
Glad you got it working