adafruit/Adafruit_nRF52_Arduino #726: No interrupt from D2?

ajs123, 2022-04-25:
PR [261](https://github.com/adafruit/Adafruit_nRF52_Arduino/pull/261) fixed issue 
[259](https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/259) which involved inability to attach interrupts to Port 1 pins. 

However, [259](https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/259) also included inability to attach an interrupt to pin D2 which is on Port 0.  Was the problem with D2 ever fixed? I am not seeing an interrupt response with attachInterrupt(2, <handler>). I do get a response on D5, which is on Port 1 and was affected by [261](https://github.com/adafruit/Adafruit_nRF52_Arduino/pull/261).

I'm working with an unmodified Feather nrf52840 Express. Per the documentation, D2, which is shared with NFC2, should be set up as a GPIO pin.

Thanks!

hathach, 2022-04-26:
I don't quite understand your question, can you provide the code along with comment on expected vs actual behavior to elaborate this issue.

ajs123, 2022-04-27:
Thanks. Yes - I'll post as a bug report.

closed 2022-04-28 by ajs123