adafruit/Adafruit_nRF52_Arduino #727: High Speed HID

rcrawfo1-brown, 2022-04-26:
**Is your feature request related to a problem? Please describe.**
For an application I'm currently working on I need to be able to send HID mouse reports consistently at a frequency of 50 Hz (20 ms between miliseconds) when connecting to an iPad (min possible connection interval 11.25 ms for HID). The function reads in a the numbers for a report from a wired serial connection and then sends out an HID packet over bluetooth. After editing the packet length and connection interval I still can't get it to this speed. Typically the function "ble.mouseReport" takes 22 ms to run, which makes it just slow enough to clog the serial port buffer and cause weird issues when running at 50 Hz. 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Looking for ways to further maximize how fast HID reports can be sent. 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
At present we just update every 40 ms, but long term we need to run at 20 ms. 

**Additional context**
Add any other context or screenshots about the feature request here.
Let me know if you need me to give additional details!

hathach, 2022-04-26:
this is application specific, there is lots of factors in minimizing ble latency, e.g connection interval, data length, tx buffer count etc.. This repo provide generic platform, for specific needs, you will need to do your own timing analyzing then tune-up and trade-off where needed (e.g ram for speed). To seek further help from user with similar need, you could post this in Adafruit support forum.

closed 2022-04-26 by hathach

tomKPZ, 2022-04-29:
I just faced this issue.  For my case the solution was to increase the queue length from the default of 1 to 256 (probably way more than necessary).  I added this code before `Bluefruit.begin();`:
```
  Bluefruit.configPrphConn(BLE_GATT_ATT_MTU_DEFAULT, BLE_GAP_EVENT_LENGTH_MIN, 256, 256);
  Bluefruit.configPrphBandwidth(BANDWIDTH_MAX);
```

pdcook, 2022-12-14:
> I just faced this issue. For my case the solution was to increase the queue length from the default of 1 to 256 (probably way more than necessary). I added this code before `Bluefruit.begin();`: [...]

You are my hero.

rcrawfo1-brown, 2022-12-14:
Above worked! It also works with 
 
> Bluefruit.configPrphConn(BLE_GATT_ATT_MTU_DEFAULT, BLE_GAP_EVENT_LENGTH_MIN, 32, 32);

and no bandwith change in my application