adafruit/Adafruit_nRF52_Arduino #728: No interrupt from pin D2

ajs123, 2022-04-27:
### Operating System

MacOS

### IDE version

Arduino CLI 0.18.3

### Board

Feather nrf52840 Express

### BSP version

Release 1.3.0

### Sketch

// Testing GPIO interrupts

#include <Arduino.h>
#include <Wire.h>

#define PIN_A 5
#define PIN_B 2

bool intA = false;
bool intB = false;

void intAHandler() {
    intA = true;
}

void intBHandler() {
    intB = true;
}

void setup() {
    Serial.begin(115200);
    while (!Serial) delay(10);
    pinMode(PIN_A, INPUT_PULLUP);
    pinMode(PIN_B, INPUT_PULLUP);
    attachInterrupt(PIN_A, intAHandler, CHANGE);
    attachInterrupt(PIN_B, intBHandler, CHANGE);
}

void loop() {
    if (intA) {
        Serial.println("Interrupt A");
        intA = false;
    }
    if (intB) {
        Serial.println("Interrupt B");
        intB = false;
    }
}

### What happened ?

Grounding pin 5 gives "Interrupt A" on the serial port, as expected.
Grounding pin 2 gives no response.

### How to reproduce ?

Load the sketch and ground either of the pins.

### Debug Log

BSP Library : 1.3.0
Bootloader  : s140 6.1.1
Serial No   : 2152F8836D2C9B52

ajs123, 2022-04-28:
I may have tracked this down to the pin setup code in cores/nRF5/nordic/nrfx/mdk/system_nrf52840.c. If CONFIG_NFCT_PINS_AS_GPIOS is not defined, then D2 remains reserved for NFC. 

For those wishing to free up the pin for GPIO, is there a preferred way to do this? I suppose that the options are

1. Change the default used at SDK compile time
2. Make it an option (in boards.txt???) 
3. Provide a function to write to the register and free up the pin

Option 1 seems wrong if some folks use NFC, or a good one if NFC is definitely not supported (and the pin protection seems pointless as long as there is no antenna).

Option 2 might be too complicated for an uncommon use case.

Option 3 would seem viable at least at least for now. I can try it and report back.

hathach, 2022-04-28:
D2 is indeed used for NFC probably by default, you could run this sketch once to convert it to normal GPIO mode

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/Hardware/nfc_to_gpio/nfc_to_gpio.ino

ajs123, 2022-04-28:
Thanks! And sorry that I missed seeing the example sketch. I use Arduino-CLI via VSCode and the examples are tucked away a bit deeper in the UI.  It seems like either the product page or the compilation options for the softdevice should change :-).

Anyway, thanks for confirming and pointing to the existing sketch.

hathach, 2022-04-30:
No problems at all, can you confirm D2 work fine after convert from nfc to gpio so that we could close this issue

ajs123, 2022-04-30:
Yes - I can confirm. I verified that I now get interrupts on D2.

hathach, 2022-05-01:
thanks for confirming

closed 2022-05-01 by hathach