adafruit/Adafruit_nRF52_Arduino #731: Cannot compile built-in example, `SerialCallResponseASCII.ino`, for `adafruit:nrf52:feather52840` with Arduino CLI

zfields, 2022-05-22:
### Operating System

Linux

### IDE version

Arduino CLI 0.22.0

### Board

Feather nRF52840 Express

### BSP version

adafruit:nrf52@1.3.0

### Sketch

File > Examples > Built-in Examples > 04.Communication/SerialCallResponseASCII

### What happened ?

The sketch references `Serial`, is able to compile, but then fails to link with `undefined reference to 'Serial'`

```
$ arduino-cli compile --build-property compiler.cpp.extra_flags='-Wno-unused-parameter -Werror' --fqbn adafruit:nrf52:feather52840:softdevice=s140v6 --warnings all ./SerialCallResponseASCII/SerialCallResponseASCII.ino
/root/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino-sketch-614CC9A98924E7946ECCBBA5233D066B/sketch/SerialCallResponseASCII.ino.cpp.o: in function `setup':
/host-volume/SerialCallResponseASCII/SerialCallResponseASCII.ino:32: undefined reference to `Adafruit_USBD_CDC::begin(unsigned long)'
/root/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /host-volume/SerialCallResponseASCII/SerialCallResponseASCII.ino:33: undefined reference to `Adafruit_USBD_CDC::operator bool()'
/root/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino-sketch-614CC9A98924E7946ECCBBA5233D066B/sketch/SerialCallResponseASCII.ino.cpp.o: in function `establishContact()':
/host-volume/SerialCallResponseASCII/SerialCallResponseASCII.ino:63: undefined reference to `Adafruit_USBD_CDC::available()'
/root/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino-sketch-614CC9A98924E7946ECCBBA5233D066B/sketch/SerialCallResponseASCII.ino.cpp.o: in function `setup':
/host-volume/SerialCallResponseASCII/SerialCallResponseASCII.ino:40: undefined reference to `Serial'
/root/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino-sketch-614CC9A98924E7946ECCBBA5233D066B/sketch/SerialCallResponseASCII.ino.cpp.o: in function `loop':
/host-volume/SerialCallResponseASCII/SerialCallResponseASCII.ino:44: undefined reference to `Adafruit_USBD_CDC::available()'
/root/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /host-volume/SerialCallResponseASCII/SerialCallResponseASCII.ino:46: undefined reference to `Adafruit_USBD_CDC::read()'
/root/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /host-volume/SerialCallResponseASCII/SerialCallResponseASCII.ino:58: undefined reference to `Serial'
collect2: error: ld returned 1 exit status

Error during build: exit status 1
```

### How to reproduce ?

1. Attempt to compile built-in example using the Arduino CLI with command, `arduino-cli compile --build-property compiler.cpp.extra_flags='-Wno-unused-parameter -Werror' --fqbn adafruit:nrf52:feather52840:softdevice=s140v6 --warnings all ./SerialCallResponseASCII/SerialCallResponseASCII.ino`
2. See linker error

### Debug Log

N/A

zfields, 2022-05-25:
I tried compiling another sketch, which led me to realize I was missing `adafruit-nrfutil`. I assumed that would fix my problem, but I was mistaken.

zfields, 2022-05-26:
I also thought it was only in Arduino CLI that caused the problem, but it turns out it's also happens in the Arduino IDE.

```
Linking everything together...
/home/zak/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/arm-none-eabi-gcc -L/tmp/arduino_build_656226 -Ofast -Wl,--gc-sections -save-temps -L/home/zak/.arduino15/packages/adafruit/hardware/nrf52/1.3.0/cores/nRF5/linker -Tnrf52840_s140_v6.ld -Wl,-Map,/tmp/arduino_build_656226/SerialCallResponseASCII.ino.map -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -u _printf_float -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--wrap=malloc -Wl,--wrap=free --specs=nano.specs --specs=nosys.specs -o /tmp/arduino_build_656226/SerialCallResponseASCII.ino.elf /tmp/arduino_build_656226/sketch/SerialCallResponseASCII.ino.cpp.o /tmp/arduino_build_656226/core/variant.cpp.o -Wl,--start-group -L/home/zak/.arduino15/packages/adafruit/tools/CMSIS/5.7.0/CMSIS/DSP/Lib/GCC/ -larm_cortexM4lf_math -lm /tmp/arduino_build_656226/../arduino_cache_769656/core/core_7d9bf707a5a0bfe0f7f1406daaaca664.a -Wl,--end-group
/home/zak/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino_build_656226/sketch/SerialCallResponseASCII.ino.cpp.o: in function `setup':
/home/zak/Development/Arduino/SerialCallResponseASCII/SerialCallResponseASCII.ino:32: undefined reference to `Adafruit_USBD_CDC::begin(unsigned long)'
/home/zak/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /home/zak/Development/Arduino/SerialCallResponseASCII/SerialCallResponseASCII.ino:33: undefined reference to `Adafruit_USBD_CDC::operator bool()'
/home/zak/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino_build_656226/sketch/SerialCallResponseASCII.ino.cpp.o: in function `establishContact()':
/home/zak/Development/Arduino/SerialCallResponseASCII/SerialCallResponseASCII.ino:63: undefined reference to `Adafruit_USBD_CDC::available()'
/home/zak/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino_build_656226/sketch/SerialCallResponseASCII.ino.cpp.o: in function `setup':
/home/zak/Development/Arduino/SerialCallResponseASCII/SerialCallResponseASCII.ino:40: undefined reference to `Serial'
/home/zak/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino_build_656226/sketch/SerialCallResponseASCII.ino.cpp.o: in function `loop':
/home/zak/Development/Arduino/SerialCallResponseASCII/SerialCallResponseASCII.ino:44: undefined reference to `Adafruit_USBD_CDC::available()'
/home/zak/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /home/zak/Development/Arduino/SerialCallResponseASCII/SerialCallResponseASCII.ino:46: undefined reference to `Adafruit_USBD_CDC::read()'
/home/zak/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /home/zak/Development/Arduino/SerialCallResponseASCII/SerialCallResponseASCII.ino:58: undefined reference to `Serial'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Adafruit Feather nRF52840 Express.
```

zfields, 2022-05-26:
I found the answer in the closed issue https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/653

The following code is required to make my sketch portable to your platform:

```
#ifdef ARDUINO_NRF52840_FEATHER
#include "Adafruit_TinyUSB.h"
#endif
```

:point_up_2:  That's fine as a workaround, but it's a poor long-term solution.

hathach, 2022-05-26:
@zfields the correct way to include tinyusb is, it is a bit annoying, but required for tinyusb to portable for other platforms: nrf52/samd/rp2040/esp32sx

```C
#ifdef USE_TINYUSB
#include "Adafruit_TinyUSB.h"
#endif

closed 2022-05-26 by hathach