adafruit/Adafruit_nRF52_Arduino #732: NRF52840 UART Problem

bromic1, 2022-05-24:
**Describe the bug:**
I cannot send data from the device to my mobile using UART.

**Set up**
- nRF52840 Sense
- Windows 11 / Arduino 2.00 RC6
- Bootloader 

**To Reproduce**
- Upload bleuart example
- Open Serial
- Start Uart in Bluefruit App
- Reset Controller
- Go to Uart in App again
- Try to send something from the device (e.g. via Serial) to the App
- nothing will be send

**Expected behavior**
- Text is shown in App

Sometimes it already doesnt work without the reset. Sending from the App to Controller always works.
Strange is that when I comment out the blebas lines it works as exprected 
When you leave Uart mode in the App and go back to Uart it also works even with blebas. But the red LED is blinking twice.
It seem that there is an interference with blebas.

hathach, 2022-05-24:
could you provide the following info for analysis
1. your mobile device: specific model + OS version
2. If you have another device, maybe try to see if it is specific to that mobile
3. the BSP Core version
4. attached your compile log as txt file
5. with un-modified bleuart example, set debug level to 1 (or 2 if needed) and post your Serial output log here as txt file
[image]

bromic1, 2022-05-26:
I found out, that the problem only occurs when the mobile phone and controller are paired/bonded. My devices were still bonded.
So maybe better take the example pairing_pin to reproduce.
When I unbond the device from mobile phone / tablet it works.
Do you still need the other information?

hathach, 2022-05-26:
@bromic1 thanks for the extra info, though I still need all the information 1-5 as mentioned in above post for reproducing.