adafruit/Adafruit_nRF52_Arduino #733: Pin names don't match pin diagram in documentation

zfields, 2022-05-25:
### Operating System

Linux

### IDE version

Arduino CLI 0.22.0

### Board

Feather nRF52840 Express

### BSP version

adafruit:nrf52@1.3.0

### Sketch

```arduino
//#define BUTTON 2 // 2 comes from the pinout diagram (attached)
#define BUTTON 0 // 0 should belong to TX based on the pinout diagram

volatile bool isr_fired = false;

void buttonISR () {
    isr_fired = true;
}

void setup () {
    ::pinMode(LED_BUILTIN, OUTPUT);
    ::pinMode(BUTTON, INPUT);
    ::attachInterrupt(digitalPinToInterrupt(BUTTON), buttonISR, RISING);
}

void loop () {
    if (isr_fired) {
        isr_fired = false;
        ::digitalWrite(LED_BUILTIN, HIGH);
        ::delay(250);
    }
    ::digitalWrite(LED_BUILTIN, LOW);
}
```

### What happened ?

With a momentary button connected to `D2`, when I define `BUTTON` as `2` the LED does not flash.
With a momentary button connected to `D2`, when I define `BUTTON` as `0` the LED flashes.

The `RX` pin can be used by defining `BUTTON` as `1` as expected.
The `TX` pin does not respond to defining `BUTTON` as `2`.

### How to reproduce ?

1. Wire up a momentary button and try using it with the pin labeled `D2`.

### Screenshots

This pinout diagramis clearly for the nRF52840 Feather Sense, but it is the only pinout diagram offered from the [nRF52840 Feather Express Guide](https://learn.adafruit.com/introducing-the-adafruit-nrf52840-feather/pinouts)

[image]

ajs123, 2022-05-28:
D2 is not available as an input (it’s configured for the NFC antenna) unless you do a one-time (and hard to reverse) reconfiguration. See [issue 728](https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/728).  

Of course, that doesn’t explain what you’re seeing when specifying pin 0.  When I encountered the issue described in 728, D0 (TX) was one of my test pins and it worked as it should.

zfields, 2022-05-29:
Where you using the **Adafruit Feather nRF52840 Express** or the **Adafruit Feather nRF52840 Sense** board?

hathach, 2022-05-30:
@zfields as mentioned by @ajs123 D2 on Feather express and sense are the same (NFC GPIO P0.10). You will need to one-time run nfc2gpio sketch to convert it from NFC to gpio
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/Hardware/nfc_to_gpio/nfc_to_gpio.ino

check out #728 for detail

zfields, 2022-05-31:
@hathach, I have a couple of questions...

1. Where did @ajs123 mention "Feather express and sense are the same"?
2. I did not run that sketch on my **Feather nRF52840 Express** and, as the issue states, it works as GPIO on pin 0. :shrug:

hathach, 2022-05-31:
Sorry if I mis-understood your description and/or didn't say it clear enough. D2 **cannot** be (reliably) used as GPIO without running the `nfc_to_gpio` sketch first. Why it responses as if it is pin0 is an accident/mystery/bug I can't explain. 

> I have not run any scripts on my Feather nRF52840 Express, it is EXACTLY as it came from the Adafruit store.

yeah, and you need to run it once if you are using stock board + bootloader for D2 to work as gpio. We did auto-convert it to gpio in later bootloader version, if you  don't want to run the sketch, just burn the latest bootloader
- https://github.com/adafruit/Adafruit_nRF52_Bootloader/blob/150a8a51f7bb02e5c54d1d5e46f19e37596fe00c/Makefile#L302

Sum up:
- To use D2 as gpio, run nfc_to_gpio or upgrade bootloader
- otherwise, skip using D2.