adafruit/Adafruit_nRF52_Arduino #734: Use Serial1 without including "Adafruit_TinyUSB.h"

arms22, 2022-06-13:
**Is your feature request related to a problem? Please describe.**

the following code causes `undefined reference to 'Adafruit_USBD_CDC::available()'` when targeting the nRF52840 boards.
I understand that the header "Adafruit_TinyUSB.h" needs to be included, but I don't want to include it for power saving.

```
void setup() {
  Serial1.begin(115200);
}

void loop() {
}
```

This is because serialEventRun refers to Serial.
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/d75bb2a8524e7f00f725eb2ab27a8996be65e7a3/cores/nRF5/Uart.cpp#L27

How about changing the call to Serial.available to a call to the weak symbol?

```
int Serial_available() __attribute__((weak));

void serialEventRun(void)
{
  if (serialEvent && Serial_available() ) serialEvent();
}
```

hathach, 2022-06-14:
I think it is better to add the `#ifdef USE_TINYUSB` for the Serial, if you make an PR. I will be happy to review/merge

arms22, 2022-06-17:
I just compiled it.
I will make PR after the test.
https://github.com/arms22/Adafruit_nRF52_Arduino/commit/1f4f503192ca4a24831a6e8ec522789c9344f78a