adafruit/Adafruit_nRF52_Arduino #737: Building for nRF52832 suddenly fails without changes when including bluefruit.h

CSC-Sendance, 2022-07-07:
### Operating System

Windows 10

### IDE version

PlatformIO Core 6.1.0 Home 3.4.2

### Board

nRF52832 Bluefruit Feather

### BSP version

release

### Sketch

```
; PlatformIO Project Configuration File

[env:adafruit_feather_nrf52832]
platform = nordicnrf52
board = adafruit_feather_nrf52832
framework = arduino

```

### What happened ?

Building a platformio project aiming with adafruit_feather_nrf52832 as target board, nordicnrf52 platform, and arduino framework , suddenly fails with a message `  #error CryptoCell CC310 is not supported on this board`   on compiling, when including <bluefruit.h>. This was not the case yesterday and also isn't on colleague's PCs. 

This part from Adafruit_nRFCrypto.cpp causes this message:
`// Only nRF52840 has CC310
#ifndef NRF_CRYPTOCELL
  #error CryptoCell CC310 is not supported on this board
#endif`

NRF_CRYPTOCELL is defined in nrf52840.h (there also appears to be no nrf52832.h) .

I would guess that this may be caused by an automatic version bump of some library but haven't found any particularly recent releases or changes in, e.g. the nrfCrypto library.

### How to reproduce ?

1. Make a new platformio project with "adafruit_feather_nrf52832" as target board
2. add `#include bluefruit.h` in main.cpp 
3. Try to build

### Debug Log

```

> Executing task: C:\.platformio\penv\Scripts\platformio.exe run --environment adafruit_feather_nrf52832 <

Processing adafruit_feather_nrf52832 (platform: nordicnrf52; board: adafruit_feather_nrf52832; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/adafruit_feather_nrf52832.html
PLATFORM: Nordic nRF52 (9.4.0) > Adafruit Bluefruit nRF52832 Feather
HARDWARE: NRF52832 64MHz, 64KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoadafruitnrf52 @ 1.10300.0 (1.3.0)
 - framework-cmsis @ 2.50700.210515 (5.7.0)
 - tool-adafruit-nrfutil @ 1.503.0 (5.3)
 - tool-sreccat @ 1.164.0 (1.64)
|   |-- Adafruit TinyUSB Library
|-- Adafruit TinyUSB Library
Building in release mode
Compiling .pio\build\adafruit_feather_nrf52832\src\main.cpp.o
Archiving .pio\build\adafruit_feather_nrf52832\libf84\libBluefruit52Lib.a
Compiling .pio\build\adafruit_feather_nrf52832\libd32\Adafruit_nRFCrypto\Adafruit_nRFCrypto.cpp.o
C:\.platformio\packages\framework-arduinoadafruitnrf52\libraries\Adafruit_nRFCrypto\src\Adafruit_nRFCrypto.cpp:36:4: error: #error CryptoCell CC310 is not supported on this board
   #error CryptoCell CC310 is not supported on this board
    ^~~~~
*** [.pio\build\adafruit_feather_nrf52832\libd32\Adafruit_nRFCrypto\Adafruit_nRFCrypto.cpp.o] Error 1
================================================================== [FAILED] Took 4.19 seconds ==================================================================
Der Terminalprozess "C:\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'adafruit_feather_nrf52832'" wurde mit folgendem Exitcode beendet: 1.

Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste.
```

hathach, 2022-07-07:
can you confirm if this issue occur with Arduino IDE. In general, we don't support PIO since we don't use it. If Arduino works then it is only a matter of PIO configure missing

CSC-Sendance, 2022-07-08:
> 

This issue does, in fact, not appear with the ArduinoIDE. 

I also tried removing the folder `C:\.platformio\packages\framework-arduinoadafruitnrf52` to force a redownload of the library, and also 'C:\.platformio\platforms\nordicnrf52'. However, the build result are the same.  So it really seems that the most up-to-date configs for platformio are buggy for the nrf52832?

Thanks for your help!

hathach, 2022-07-08:
since it works with Arduino IDE, it is probably issue with PIO configuration. Unfortunately, I don't use PIO and have no idea how its configuration works, therefore couldn't help you at all with this. I think you are best to post this issue on PIO forum, there is lots of people could help you there.

hathach, 2022-07-11:
resolved.

closed 2022-07-11 by hathach