adafruit/Adafruit_nRF52_Arduino #741: Add .non_init RAM

ericlangel, 2022-08-10:
Is it (already) possible to use variables that are not initialised at softreset?
like:
static uint8_t data[16] __attribute__((section(".non_init")));
or

So after a power-On Reset the Variable "counter" is random or zero
the counter is increased by application.

after a softreset the variable "counter" should contain the old value before the softreset

Reason: i have an Application running which does a reset from time to time (i think it happens at BLE transfer) and some values should be hold. and i cant writ them in Filesystem, because i dont know the time of the next softreset.

brailovskii, 2023-07-31:
Could you solve that problem? 
I think linker file should be modified by adding non_init section

ericlangel, 2023-08-03:
no i couldnt solve the non_init problem

but i solved my Problem with the Soft reset. So i didnt followed this issue

so for me it is more or less closed
but there is no real solution for the issue