adafruit/Adafruit_nRF52_Arduino #742: Extraordinary power consumption when using attachInterrupt and Wire (I2C)

jgartrel, 2022-09-13:
### Operating System

MacOS

### IDE version

1.8.8

### Board

Custom board based on nrf52832

### BSP version

1.3.0

### Sketch

The code snippit below seems to be the simplest way to express the issue:
```c
#include <bluefruit.h>
#include <Wire.h> //I2C library

#define INT_PIN 15

volatile unsigned long ms = 25;

void irqHandler()
{
  #if CFG_SYSVIEW
  SEGGER_SYSVIEW_RecordEnterISR();
  #endif

  ms = 2000;

  #if CFG_SYSVIEW
  SEGGER_SYSVIEW_RecordExitISR();
  #endif
}

void setup() {
  Bluefruit.begin();
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, !LED_STATE_ON);

  Wire.begin();
  delay(100);

  Wire.beginTransmission(0x0E);
  Wire.write(0x0F);
  Wire.endTransmission();

  Wire.end();
  delay(100);

  pinMode(INT_PIN, INPUT_PULLUP);
  attachInterrupt(INT_PIN, irqHandler, FALLING);
}

void loop() 
{
  digitalWrite(LED_BUILTIN, LED_STATE_ON);
  delay(ms);
  digitalWrite(LED_BUILTIN, !LED_STATE_ON);
  delay(10000);
}
```

### What happened ?

When using BOTH the Wire Library (I2C) and attachInterrupt,  power consumption increases dramatically from <11uA to 472uA.

### How to reproduce ?

I seem to have a similar issue to that which @ericlangel _originally posted in https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/165#issuecomment-1015328540_

Testing with the nrf52832:  When using BOTH the Wire Library (I2C) and ```attachInterrupt```,  power consumption increases dramatically from <11uA to 472uA.

I ran 4 tests and measured the idle current during the ```delay(10000)``` in loop:

1.  Baseline (No I2C, no attachInterrupt) - 2.525uA
2.  attachInterrupt only - 10.25uA
3.  I2C only - 2.525uA
4.  Both I2C and attachInterrupt - 472uA (**I expect to see 10.25uA here**)

These tests can be reproduced by modifying the above posted source in the following ways:

Test 1 - (2.525uA):
```c
//  Wire.beginTransmission(0x0E);
//  Wire.write(0x0F);
//  Wire.endTransmission();
...
//  attachInterrupt(INT_PIN, irqHandler, FALLING);
```

Test 2 - (10.25uA):
```c
//  Wire.beginTransmission(0x0E);
//  Wire.write(0x0F);
//  Wire.endTransmission();
...
  attachInterrupt(INT_PIN, irqHandler, FALLING);
```

Test 3 - (2.525uA):
```c
  Wire.beginTransmission(0x0E);
  Wire.write(0x0F);
  Wire.endTransmission();
...
//  attachInterrupt(INT_PIN, irqHandler, FALLING);
```

Test 4 - (472uA):
```c
  Wire.beginTransmission(0x0E);
  Wire.write(0x0F);
  Wire.endTransmission();
...
  attachInterrupt(INT_PIN, irqHandler, FALLING);
```

ericlangel, 2022-12-21:
any news here?

jgartrel, 2023-02-21:
I repeated the tests above on the [Adafruit Feather nRF52 Bluefruit LE - nRF52832](https://www.adafruit.com/product/3406) board. I confirmed that this issue will appear on all boards based on the nRF52832 and likely affects all nRF52 boards using the v1.3.0 of this BSP.

All tests were done by using the nRF-PPK2 sourcing 3.7V connected to VBAT on the Feather nrf52832.

Results were as follows (using the same code and tests listed above):

Test 1 - (114uA average, 258uA peak):
```C
//  Wire.beginTransmission(0x0E);
//  Wire.write(0x0F);
//  Wire.endTransmission();
...
//  attachInterrupt(INT_PIN, irqHandler, FALLING);
```

Test 2 - (119uA average, 247uA peak)
```C
//  Wire.beginTransmission(0x0E);
//  Wire.write(0x0F);
//  Wire.endTransmission();
...
  attachInterrupt(INT_PIN, irqHandler, FALLING);
```

- Test 3 - (123uA average,   262uA peak)
```C
  Wire.beginTransmission(0x0E);
  Wire.write(0x0F);
  Wire.endTransmission();
...
//  attachInterrupt(INT_PIN, irqHandler, FALLING);
```

- Test 4 -  (**643uA** average, 3.95mA peak) - this is more than 515uA outside the expected range of 114uA - 128uA
```C
  Wire.beginTransmission(0x0E);
  Wire.write(0x0F);
  Wire.endTransmission();
...
  attachInterrupt(INT_PIN, irqHandler, FALLING);
```

jgartrel, 2024-06-08:
any news on this bug?

ion-popa, 2025-07-27:
@jgartrel There is a bug on the die level for nRF52832 I2C module. To solve the issue, you have to call `Wire.end();` after using the Wire class.

jgartrel, 2025-08-12:
@ion-popa Even with calling Wire.end();  (see the original description of this issue), we still see extraordinary power consumption when using attachInterrupt after making some Wire (I2C) calls.