adafruit/Adafruit_nRF52_Arduino #743: Unicode/locale error on MacOS and Arduino 2.0.0

exipilis, 2022-10-04:
### Operating System

MacOS

### IDE version

2.0.0

### Board

Feather nRF52840 Sense

### BSP version

1.3.0

### Sketch

Blink

### What happened ?

With Arduino IDE 1.8.x nRF52 boards compile successfully.
With Arduino IDE 2.0.0 ARM and AVR boards compile successfully.
With Arduino IDE 2.0.0 and nRF52 I get this while compiling the sketch:

Traceback (most recent call last):
  File "__main__.py", line 317, in <module>
  File "click/core.py", line 1134, in __call__
  File "click/core.py", line 1040, in main
  File "click/_unicodefun.py", line 100, in _verify_python_env
RuntimeError: Click will abort further execution because Python was configured to use ASCII as encoding for the environment. Consult https://click.palletsprojects.com/unicode-support/ for mitigation steps.

This system lists some UTF-8 supporting locales that you can pick from. The following suitable locales were discovered: af_ZA.UTF-8, am_ET.UTF-8, be_BY.UTF-8, bg_BG.UTF-8, ca_ES.UTF-8, cs_CZ.UTF-8, da_DK.UTF-8, de_AT.UTF-8, de_CH.UTF-8, de_DE.UTF-8, el_GR.UTF-8, en_AU.UTF-8, en_CA.UTF-8, en_GB.UTF-8, en_IE.UTF-8, en_NZ.UTF-8, en_US.UTF-8, es_ES.UTF-8, et_EE.UTF-8, eu_ES.UTF-8, fi_FI.UTF-8, fr_BE.UTF-8, fr_CA.UTF-8, fr_CH.UTF-8, fr_FR.UTF-8, he_IL.UTF-8, hr_HR.UTF-8, hu_HU.UTF-8, hy_AM.UTF-8, is_IS.UTF-8, it_CH.UTF-8, it_IT.UTF-8, ja_JP.UTF-8, kk_KZ.UTF-8, ko_KR.UTF-8, lt_LT.UTF-8, nl_BE.UTF-8, nl_NL.UTF-8, no_NO.UTF-8, pl_PL.UTF-8, pt_BR.UTF-8, pt_PT.UTF-8, ro_RO.UTF-8, ru_RU.UTF-8, sk_SK.UTF-8, sl_SI.UTF-8, sr_YU.UTF-8, sv_SE.UTF-8, tr_TR.UTF-8, uk_UA.UTF-8, zh_CN.UTF-8, zh_HK.UTF-8, zh_TW.UTF-8
[4418] Failed to execute script __main__

exit status 1

Compilation error: exit status 1

Yes, I tried to consult https://click.palletsprojects.com/unicode-support/. It did not help.

### How to reproduce ?

Open any sketch.
Click Verify.

ApolloDS, 2023-01-10:
Also check the issue here:
[Unicode/Locale compilation error #1513](https://github.com/arduino/arduino-ide/issues/1513)

per1234, 2023-01-18:
I am not able to reproduce the fault with versions of this platform that use **adafruit-nrfutil** 0.5.3.post16, which was introduced at https://github.com/adafruit/Adafruit_nRF52_Arduino/commit/19ee2ba5f305bd69cb0257dd3b83d1eb5521c65b ("**Adafruit nRF52**" platform version 0.22.1 or newer if you are using Boards Manager).

I am able to reproduce the fault with versions that use **adafruit-nrfutil** 0.5.3.post12 before the update at https://github.com/adafruit/Adafruit_nRF52_Arduino/commit/19ee2ba5f305bd69cb0257dd3b83d1eb5521c65b ("**Adafruit nRF52**" platform 0.22.0 or older if you are using Boards Manager).

I'm wondering why my results don't match with this part of the issue report:

> ### BSP version
> 1.3.0

(I can't reproduce the fault using "**Adafruit nRF52**" platform version 1.3.0)

@exipilis please post the verbose compilation output so I can see the exact command that produced the error:

1. Select **File > Preferences** from the Arduino IDE menus.
1. Check the box next to "**Show verbose output during: > ☐ compilation**"
1. Click the **<kbd>OK</kbd>** button.
1. Select **File > New Sketch** from the Arduino IDE menus.
1. Select **Tools > Board > Adafruit nRF52 > Feather nRF52840 Sense** from the Arduino IDE menus.
1. Select **Sketch > Verify/Compile** from the Arduino IDE menus.
1. Wait for the compilation to fail.
1. You will see a "**Compilation error: ...**" notification at the bottom right corner of the Arduino IDE window. Click the <kbd>**COPY ERROR MESSAGES**</kbd> button on that notification.
1. [Paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the error messages in a reply here.
   In order to ensure the text is not corrupted, please use [code fencing](https://guides.github.com/features/mastering-markdown/).

Mona-Roza, 2023-07-21:
Traceback (most recent call last):
  File "__main__.py", line 317, in <module>
  File "click/core.py", line 1134, in __call__
  File "click/core.py", line 1040, in main
  File "click/_unicodefun.py", line 100, in _verify_python_env
RuntimeError: Click will abort further execution because Python was configured to use ASCII as encoding for the environment. Consult https://click.palletsprojects.com/unicode-support/ for mitigation steps.

This system lists some UTF-8 supporting locales that you can pick from. The following suitable locales were discovered: af_ZA.UTF-8, am_ET.UTF-8, be_BY.UTF-8, bg_BG.UTF-8, ca_ES.UTF-8, cs_CZ.UTF-8, da_DK.UTF-8, de_AT.UTF-8, de_CH.UTF-8, de_DE.UTF-8, el_GR.UTF-8, en_AU.UTF-8, en_CA.UTF-8, en_GB.UTF-8, en_IE.UTF-8, en_NZ.UTF-8, en_US.UTF-8, es_ES.UTF-8, et_EE.UTF-8, eu_ES.UTF-8, fi_FI.UTF-8, fr_BE.UTF-8, fr_CA.UTF-8, fr_CH.UTF-8, fr_FR.UTF-8, he_IL.UTF-8, hr_HR.UTF-8, hu_HU.UTF-8, hy_AM.UTF-8, is_IS.UTF-8, it_CH.UTF-8, it_IT.UTF-8, ja_JP.UTF-8, kk_KZ.UTF-8, ko_KR.UTF-8, lt_LT.UTF-8, nl_BE.UTF-8, nl_NL.UTF-8, no_NO.UTF-8, pl_PL.UTF-8, pt_BR.UTF-8, pt_PT.UTF-8, ro_RO.UTF-8, ru_RU.UTF-8, sk_SK.UTF-8, sl_SI.UTF-8, sr_YU.UTF-8, sv_SE.UTF-8, tr_TR.UTF-8, uk_UA.UTF-8, zh_CN.UTF-8, zh_HK.UTF-8, zh_TW.UTF-8
[5993] Failed to execute script __main__

exit status 1

Compilation error: exit status 1

mzks, 2023-08-22:
I faced the same error for Xiao BLE nRF52840 with the following instruction https://wiki.seeedstudio.com/XIAO_BLE/.
My environment is M1 Max MacBook Pro and OS version is 13.4.1.
Arduino IDE version is 2.1.1.

Traceback (most recent call last):
  File "__main__.py", line 317, in <module>
  File "click/core.py", line 1134, in __call__
  File "click/core.py", line 1040, in main
  File "click/_unicodefun.py", line 100, in _verify_python_env
RuntimeError: Click will abort further execution because Python was configured to use ASCII as encoding for the environment. Consult https://click.palletsprojects.com/unicode-support/ for mitigation steps.

This system lists some UTF-8 supporting locales that you can pick from. The following suitable locales were discovered: af_ZA.UTF-8, am_ET.UTF-8, be_BY.UTF-8, bg_BG.UTF-8, ca_ES.UTF-8, cs_CZ.UTF-8, da_DK.UTF-8, de_AT.UTF-8, de_CH.UTF-8, de_DE.UTF-8, el_GR.UTF-8, en_AU.UTF-8, en_CA.UTF-8, en_GB.UTF-8, en_IE.UTF-8, en_NZ.UTF-8, en_US.UTF-8, es_ES.UTF-8, et_EE.UTF-8, eu_ES.UTF-8, fi_FI.UTF-8, fr_BE.UTF-8, fr_CA.UTF-8, fr_CH.UTF-8, fr_FR.UTF-8, he_IL.UTF-8, hr_HR.UTF-8, hu_HU.UTF-8, hy_AM.UTF-8, is_IS.UTF-8, it_CH.UTF-8, it_IT.UTF-8, ja_JP.UTF-8, kk_KZ.UTF-8, ko_KR.UTF-8, lt_LT.UTF-8, nl_BE.UTF-8, nl_NL.UTF-8, no_NO.UTF-8, pl_PL.UTF-8, pt_BR.UTF-8, pt_PT.UTF-8, ro_RO.UTF-8, ru_RU.UTF-8, sk_SK.UTF-8, sl_SI.UTF-8, sr_YU.UTF-8, sv_SE.UTF-8, tr_TR.UTF-8, uk_UA.UTF-8, zh_CN.UTF-8, zh_HK.UTF-8, zh_TW.UTF-8
[26979] Failed to execute script __main__

exit status 1

Compilation error: exit status 1

mzks, 2023-08-22:
Arduino legacy (version 1.8.19) works well.

hathach, 2023-11-30:
not reproducible with latest Arduino BSP Core: 1.5.0 and IDE 2.2.1. Please try to update both BSP and IDE to latest and give it a try again. Feel free to comment, re-open issue should it still exists with latest BSP & IDE.

closed 2023-11-30 by hathach

ingmanuelalfonso, 2024-01-04:
h

finally-fancy, 2024-01-07:
I have the smae error with the latest IDE 2.2.1 

I'm not sure which python version the IDE ais using. how to analyse this?. I have set the defalt local in my .zshrc and my normal python dev env is working well with the locale

LucScu, 2024-01-08:
@finally-fancy try to add 
export LC_ALL=en_US.utf-8
export LANG=en_US.utf-8
to the root user .profile file in /var/root

jefflai0101, 2024-01-21:
> I faced the same error for Xiao BLE nRF52840 with the following instruction https://wiki.seeedstudio.com/XIAO_BLE/. My environment is M1 Max MacBook Pro and OS version is 13.4.1. Arduino IDE version is 2.1.1.
> 
> Traceback (most recent call last): File "**main**.py", line 317, in File "click/core.py", line 1134, in **call** File "click/core.py", line 1040, in main File "click/_unicodefun.py", line 100, in _verify_python_env RuntimeError: Click will abort further execution because Python was configured to use ASCII as encoding for the environment. Consult https://click.palletsprojects.com/unicode-support/ for mitigation steps.
> 
> This system lists some UTF-8 supporting locales that you can pick from. The following suitable locales were discovered: af_ZA.UTF-8, am_ET.UTF-8, be_BY.UTF-8, bg_BG.UTF-8, ca_ES.UTF-8, cs_CZ.UTF-8, da_DK.UTF-8, de_AT.UTF-8, de_CH.UTF-8, de_DE.UTF-8, el_GR.UTF-8, en_AU.UTF-8, en_CA.UTF-8, en_GB.UTF-8, en_IE.UTF-8, en_NZ.UTF-8, en_US.UTF-8, es_ES.UTF-8, et_EE.UTF-8, eu_ES.UTF-8, fi_FI.UTF-8, fr_BE.UTF-8, fr_CA.UTF-8, fr_CH.UTF-8, fr_FR.UTF-8, he_IL.UTF-8, hr_HR.UTF-8, hu_HU.UTF-8, hy_AM.UTF-8, is_IS.UTF-8, it_CH.UTF-8, it_IT.UTF-8, ja_JP.UTF-8, kk_KZ.UTF-8, ko_KR.UTF-8, lt_LT.UTF-8, nl_BE.UTF-8, nl_NL.UTF-8, no_NO.UTF-8, pl_PL.UTF-8, pt_BR.UTF-8, pt_PT.UTF-8, ro_RO.UTF-8, ru_RU.UTF-8, sk_SK.UTF-8, sl_SI.UTF-8, sr_YU.UTF-8, sv_SE.UTF-8, tr_TR.UTF-8, uk_UA.UTF-8, zh_CN.UTF-8, zh_HK.UTF-8, zh_TW.UTF-8 [26979] Failed to execute script **main**
> 
> exit status 1
> 
> Compilation error: exit status 1

Hi, I am under similar environment and same chip.

My problem was caused by adafruit-nrfutil. I found a way to workaround for my side.

I installed the adafruit-nrfutil with pip, and copied the pip version to where the adafruit-nrfutil is causing problem:

`pip install adafruit-nrfutil`
`mv "original-path/adafruit-nrfutil" "original-path/adafruit-nrfutil_bak"`
`cp "pip-version/adafruit-nrfutil" "original-path/"`

I hope this helps.

oleander, 2024-01-22:
Installing `adafruit-nrfutil` from elsewhere and overriding the one used by Arduino IDE worked for me.

```
# pip3 install --user adafruit-nrfutil
find ~/Library/Arduino15/packages/adafruit/hardware/nrf52/* -name adafruit-nrfutil -exec ln -sf $(which adafruit-nrfutil) {} \;
```

todocono, 2024-11-20:
Thanks, @Steppschuh . That did the trick for me. I copied the file adafruit-nrfutil within the Seeduino board definition, as you said:

- Installed both boards from Seeed: nrf52 and nrf52 mbed-enabled
- Both files are in the same directory /Users/yourUser/Arduino15/packages/Seeeduino/hardware
- The file in the older mbed folder inside of 2.9.2/tools/adafruit-nrfutil/macos is the one working
- You should put that file in the newer nrf52 folder inside of 1.1.8/tools/adafruit-nrfutil/macos

e3e6, 2025-05-15:
For MacOS:

Execute
```
nano ~/.zshrc
```
OR
```
nano ~/.bash_profile
```
Add these two lines at the end:
```
export LC_ALL=en_US.UTF-8
export LANG=en_US.UTF-8
```
Save and exit (in nano: `press Ctrl + X`, then `Y`, then `Enter`).
Reload your config:
```
source ~/.zshrc
```
OR
```
source ~/.bash_profile
```
Start Arduino IDE

hathach, 2025-08-07:
re-open, it seems to be macos specific issue.