adafruit/Adafruit_nRF52_Arduino #762: SPI hardware issue(?) - MISO held low crashes nrf52

ColemanMerchant, 2023-03-06:
### Operating System

MacOS

### IDE version

1.8.19

### Board

nrf52840 Express

### BSP version

1.3.0

### Sketch

Blink Example / DigitalReadSerial (added: #include <Adafruit_TinyUSB.h> // Needed for NRF52840 USB CDC)

### What happened ?

When hardware MISO input (pin 24) is connected to ground, sketches will run properly after upload, but then if the board is power cycled by unplugging the USB and re-plugging it in, the serial port connects but nothing further happens (board seems to hang). This is true for any sketch, even the most basic (just a serial.print). The board restarts and works as normal if MISO is left floating.

Background: I'm working with an unusual SPI sensor (Heimann thermopile array) that uses its only CS input to switch between EEPROM and the SENSOR (respectively high and low), so it is never fully deactivated and must be the only device on the bus (quite annoying). The sensor works fine like this, but only immediately after flashing code - if power is cycled, the nrf52 hangs (as replicated above), seemingly due to MISO line being held low.

I have tried on M4 express, M0 express and M0 Bluefruit - all do not exhibit this behavior.

### How to reproduce ?

1. Flash any sketch while MISO (pin 24) is held low.
2. Cycle power with USB cable.

hathach, 2023-11-30:
not reproducible, I have tested with Feather nRF52840 Express and it works just fine with blinky sketch with MISO pin wiring to GND. Maybe trying to update BSP core and libraries, IDE and try again, if you still have the issue, please take the picture of your board setup, so that we could examine what is wrong.

ColemanMerchant, 2023-11-30:
Thanks for taking a look at this! It seems it is actually going into DFU bootloader mode when MISO is held low during power-up. I updated IDE and BSP core to 1.5.0, and that had no effect. Reproducible for me on multiple unconnected/bare boards.

My fix was to put the (errant) SPI device on an alternate set of pins and that fixed the issue for my specific application.

https://github.com/adafruit/Adafruit_nRF52_Arduino/assets/74724400/aef6f0c4-3260-463c-9a92-2f0caea999e8