adafruit/Adafruit_nRF52_Arduino #765: Ble UART code only works with android phones but not on iphones

SudharsanK58, 2023-03-30:
### Operating System

Windows 10

### IDE version

Platform IO

### Board

nRF52 Dev Kit

### BSP version

Latest version of blefruit lib

### Sketch

#include <bluefruit.h>

BLEDfu  bledfu;  // OTA DFU service
BLEDis  bledis;  // device information
BLEUart bleuart; // uart over ble
BLEBas  blebas;  // battery

static void startAdv(void);
static void connect_callback(uint16_t conn_handle);
static void disconnect_callback(uint16_t conn_handle, uint8_t reason);

void setup()
{
  Serial.begin(115200);
  while ( !Serial ) delay(10);  
  
  Serial.println("Bluefruit52 BLEUART Example");
  Serial.println("---------------------------\n");

  Bluefruit.autoConnLed(true);

  Bluefruit.configPrphBandwidth(BANDWIDTH_MAX);

  Bluefruit.begin();
  Bluefruit.setTxPower(4);    // Check bluefruit.h for supported values
  Bluefruit.setName("Bluefruit52");
  Bluefruit.Periph.setConnectCallback(connect_callback);
  Bluefruit.Periph.setDisconnectCallback(disconnect_callback);

  bledfu.begin();

  bledis.setManufacturer("Adafruit Industries");
  bledis.setModel("Bluefruit Feather52");
  bledis.begin();

  bleuart.begin();

  blebas.begin();
  blebas.write(100);

  startAdv();

  Serial.println("Please use Adafruit's Bluefruit LE app to connect in UART mode");
  Serial.println("Once connected, enter character(s) that you wish to send");
}

void startAdv(void)
{
  Bluefruit.Advertising.addFlags(BLE_GAP_ADV_FLAGS_LE_ONLY_GENERAL_DISC_MODE);
  Bluefruit.Advertising.addTxPower();

  Bluefruit.Advertising.addService(bleuart);

  Bluefruit.ScanResponse.addName();
  
  Bluefruit.Advertising.restartOnDisconnect(true);
  Bluefruit.Advertising.setInterval(32, 244);    // in unit of 0.625 ms
  Bluefruit.Advertising.setFastTimeout(30);      // number of seconds in fast mode
  Bluefruit.Advertising.start(0);                // 0 = Don't stop advertising after n seconds  
}

void loop()
{
  while (Serial.available())
  {
    delay(2);

    uint8_t buf[64];
    int count = Serial.readBytes(buf, sizeof(buf));
    bleuart.write( buf, count );
  }

  while ( bleuart.available() )
  {
    uint8_t ch;
    ch = (uint8_t) bleuart.read();
    Serial.write(ch);
  }
}

void connect_callback(uint16_t conn_handle)
{
  BLEConnection* connection = Bluefruit.Connection(conn_handle);

  char central_name[32] = { 0 };
  connection->getPeerName(central_name, sizeof(central_name));

  Serial.print("Connected to ");
  Serial.println(central_name);
}

void disconnect_callback(uint16_t conn_handle, uint8_t reason)
{
  (void) conn_handle;
  (void) reason;

  Serial.println();
  Serial.println("Disconnected");
}

### What happened ?

Its a simple ble UART example. This codes receives UART data for android phones only. But not on IOS iphones why?

### How to reproduce ?

1.Run this code on Platform IO or Arduino IDE.
2.Ble named "Bluefruit52" will be shown on UART app or NrF connect app.
3.Try to send data via UART or try to connect with iphone.

mhbkamaei, 2023-06-16:
Since iOS 13.3 (?) devices with Legacy DFU that do not use bonding, but use direct advertising in DFU Bootloader mode cannot be updated anymore. The same applies to Android devices with latest security patches. The solution is to modify the bootlaoder on a device to use non-direct advertising, but in order to update devices that are already in production, older, not updated devices are required. Nothing can be done on the library side, as the iOS just stopped reporting such advertising packets.

https://github.com/NordicSemiconductor/IOS-DFU-Library/blob/main/documentation.md

SudharsanK58, 2023-06-19:
I got it , Thanks a lot @mhbkamaei

closed 2023-11-30 by hathach