adafruit/Adafruit_nRF52_Arduino #768: What Windows Driver to Use on Raytac nrf52840 TinyUSB Dongle

ispybadguys, 2023-04-06:
### Operating System

Windows 10

### IDE version

2.04

### Board

Raytac nrf52840 TinyUSB Dongle

### BSP version

latest

### Sketch

NA

### What happened ?

Windows sees USB device but doesn't load as 210x driver and I can't talk to the USB. Sketch work fine on Mac.

### How to reproduce ?

Attach device to USB

### Debug Log

None

### Screenshots

NA

hathach, 2023-04-07:
nrf52840 does not use cp210x, please install adafruit driver to use with acm on raytac dongle https://github.com/adafruit/Adafruit_Windows_Drivers/releases

dhalbert, 2023-04-07:
You should not need to install any driver on Windows 10. There is no specific support for the Raytac dongle in Adafruit Windows driver release. Did you buy https://www.adafruit.com/product/5199 ? That product has the UF2 bootloader installed. If you got the dongle somewhere else, I believe it comes with Nordic's standard bootloader.

ispybadguys, 2023-04-07:
I did get the Adafruit Product  https://www.amazon.com/dp/B09DMK445K?psc=1&ref=ppx_yo2ov_dt_b_product_details. I have no issue with the device programming it on a MacBook Pro using Arduino IDE 2.06 but when I take it to a Windows 10 machine I cannot not communicate with it using the IDE Serial. It creates a USB COM device and I can open it with the Serial Monitor but that is it.

dhalbert, 2023-04-07:
I think this is a support issue probably having to do with your Windows installation. We can move this discussion to https://forums.adafruit.com, or you could get real-time community-based support in our discord at https:adafru.it/discord.

dhalbert, 2023-04-07:
Please ping me here if you open a Forums thread or post on discord.

closed 2023-04-07 by dhalbert

dhalbert, 2023-04-07:
Note that the way to get the BOOT drive to show is to hold down the button on the dongle and plug it in. It does not work to double-click the button.

ispybadguys, 2023-04-07:
Thanks Dan. Maybe move to forums.  I wrote the sketch to talk to an ASCOM driver i.e. over USB Serial.. I discovered that the driver has no problem opening the device and communicating with it but the IDE,2.0 version or version 18 , cannot open it and talk to it. The serial monitor opens but no output is seen. I checked and the Baud rate is correct. Serial.begin never returns true with the serial monitor. Works fin on a Mac.