adafruit/Adafruit_nRF52_Arduino #770: FastLED Crashing

someburner, 2023-05-20:
### Operating System

Linux

### IDE version

platformio

### Board

Feather nRF2840, bluefruit 2840

### BSP version

latest

### Sketch

literally any `FastLED.show()` call crashes almost immediately.

### What happened ?

Remove FastLED, or just dont' call `show()` and application runs perfectly.

### How to reproduce ?

Call `FastLED.show()`.

hathach, 2023-05-24:
As far as I know It is implemented already https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/109. Please try to ask or file an issue on fastLED since they know their code better. You could open an issue with specific changes request if it is narrow downed to be the specific issue of this core PWM implemenation.

closed 2023-05-24 by hathach

someburner, 2023-05-24:
Sounds good. I just thought you guys might have some ideas since FastLED doesn't seem very active and this would be a nice benefit for those of us that enjoy purchasing adafruit hardware. Neopixel works so I just am thinking there are some tricks that need to be applied to FastLED or maybe a more recent nordic toolchain broke something.

hathach, 2023-05-25:
we would like to help but not familliar with FastLED codebase and usage. I am sure there is other user running it on nrf52 in the FastLED dicussion/forum. Once posted in FastLed repo, you could link to this issue, I will also try to follow to see if there is anything I could help there.

someburner, 2023-06-03:
@hathach 

Submitted an issue over there. I agree that is the best place for it, but I am wondering if you (or anyone looking?) could simply give it a try yourself with latest `Adafruit_nRF52_Arduino` and latest `FastLED` just so I can figure out if this is easy for others to reproduce, or if I'm just missing something stupid.

For me the issue is very easy to reproduce, just hook up a WS2812B strip of any length (mine is 64 leds, but doesnt seem to matter), call the FastLED setup code, do `FastLED.show()`, and see if it locks up.

someburner, 2023-06-04:
Turns out it was due to platformio optimizing out the ISR handler. -_-

hathach, 2023-06-05:
ah great, thank you for your update. I am sure someone with same issue will find this info helpful. Yeah, platformio can do more optimization and neat trick that is why I usually ask people to double check with Arduino IDE and then try to compare with Arduino first.