adafruit/Adafruit_nRF52_Arduino #771: I2C communication between NRF52832 hangs

ssysm, 2023-05-24:
### Operating System

MacOS

### IDE version

Arduino 2.1.0

### Board

Feather 52832

### BSP version

1.4.0

### Sketch

Receiver: 

```
#include <Wire.h>
#include <Adafruit_TinyUSB.h>

void setup()
{
  Serial.begin(115200);
  while(! Serial) delay(10);
  Wire.begin(0x50);
  Wire.onReceive(receiveEvent);
}

void loop()
{
  delay(100);
}

// function that executes whenever data is received from master
// this function is registered as an event, see setup()
void receiveEvent(int howMany)
{
  while(1 < Wire.available()) // loop through all but the last
  {
    char c = Wire.read(); // receive byte as a character
    Serial.print(c);         // print the character
  }
  int x = Wire.read();    // receive byte as an integer
  Serial.println(x);         // print the integer
}
```

Sender: 

```
#include <Wire.h>

void setup()
{
  Wire.begin(); // join i2c bus (address optional for main)
}

byte x = 0;

void loop()
{
  Wire.beginTransmission(0x50); // transmit to device #0x50
  Wire.write("x is ");        // sends five bytes
  Wire.write(x);              // sends one byte  
  Wire.endTransmission();    // stop transmitting

  x++;
  delay(500);
}
```

### What happened ?

Two NRF52832 Feather is connected via I2C port. When sender is running without the receiver attached, the sketch is running fine. But when the receiver is attached, the sender hangs on endTramission.

### How to reproduce ?

Connect Sender SCL, SDA to another NRF52832 SCL and SDA port. Connect a 2.2k Pull up resistor between SCL and VCC, SDA and VCC. Once receiver is attached, the sender will hang.

### Screenshots

I2C bus scope capture: 
[image]

apoorv1in, 2024-06-07:
I see the Wire_nRF52.cpp does not have timeout implemented endTransmission() and requestFrom()
there are multiple while loops and none have timeout implemented. Under certain scenarios then the call never returns.

If i take reference from Arduino library it has the timeout https://github.com/arduino/ArduinoCore-avr/blob/master/libraries/Wire/src/utility/twi.c

JTerminator, 2025-02-16:
Had a similar problem, and adding a basic timeout to the Wire_nRF52.cpp "solved" my problem as in my programm doesn't flat out crash anymore on singular unsuccessful I2C-Writes. When I have more time, I should investigate on why the the communication gets stuck. But I really think a timeout should be added nonetheless.

strasharo, 2026-08-31:
Hitting what looks like the same root cause @apoorv1in identified, but as a **boot-time freeze** rather than a mid-operation hang, on different hardware (Seeed XIAO nRF52840 + Wio-SX1262, via the [MeshCore](https://github.com/meshcore-dev/MeshCore) firmware, which vendors this core unmodified).

MeshCore's boot sequence runs an RTC autodetection scan (`i2c_probe()`: `beginTransmission(addr)` / `endTransmission()` against a handful of hardcoded RTC chip addresses) before anything else initializes. On this board, with no RTC or sensor actually attached to the I2C pins, that scan permanently hangs the device on every cold boot — no serial output, no BLE advertising, nothing. Confirmed across firmware v1.10.0 through current (see [meshcore-dev/MeshCore#2068](https://github.com/meshcore-dev/MeshCore/issues/2068)).

Traced it to the same three unbounded `while(!EVENT)` loops in `endTransmission()`:

```cpp
_p_twim->TASKS_STARTTX = 0x1UL;
while(!_p_twim->EVENTS_TXSTARTED && !_p_twim->EVENTS_ERROR);   // #1
...
while(!_p_twim->EVENTS_LASTTX && !_p_twim->EVENTS_ERROR);      // #2
...
_p_twim->TASKS_STOP = 0x1UL;
while(!_p_twim->EVENTS_STOPPED);                                // #3 - no EVENTS_ERROR escape at all
```

Loop #3 in particular has no error-escape clause whatsoever, so even in the two prior loops' error path, the code still falls through to a `while` that can never return if `EVENTS_STOPPED` doesn't fire.

Given this now has two independent reports (different boards, different call sites — one mid-transaction, one at boot) landing on the identical unbounded-loop cause, seems worth prioritizing a timeout fix (matching what `ArduinoCore-avr`'s `twi.c` does, as you linked) over leaving it to each downstream project to work around by deleting their I2C probe calls, which is what MeshCore users are currently doing.

Happy to test a patch against this exact board+radio combo if one gets proposed.