adafruit/Adafruit_nRF52_Arduino #773: SPI requires VBUS to work

amowry, 2023-06-08:
### Operating System

Windows 11

### IDE version

2.1.0

### Board

ItsyBitsy NRF52840 Express

### BSP version

release

### Sketch

Any SPI sketch

### What happened ?

SPI doesn't work if VBUS is not present at startup. It is the same as this SPIM3 bug in the CircuitPython version, but I thought I'd mention it here because I'd like to know if there is a workaround or fix for the Arduino environment:
https://github.com/adafruit/circuitpython/issues/5233

Also discussed here:
https://forums.adafruit.com/viewtopic.php?t=188798&start=15

And here: 
https://devzone.nordicsemi.com/f/nordic-q-a/83266/spim3-not-work-standalone/350598

### How to reproduce ?

Use SPI without VBUS present (e.g. with battery power)
MOSI never fires.

dhalbert, 2023-06-08:
We have been unable to get a response from Noridc about this problem. It appears to be a hardware issue, or an undocumented issue with the Nordic driver that should handle this case. As you see, there has been no response in the Nordic DevZone about this. SPIM3 has had hardware issues. You might raise the issue yourself in DevZone, perhaps opening a new issue. An Arduino example might get further than our CircuitPython observations, though they may want an nRF SDK or nrfx example.

amowry, 2023-06-08:
Thanks for the info and quick response! I can force it to use SPIM2 instead of SPIM 3 by changing this in SPI.cpp:

`// default to 0
#ifndef SPI_32MHZ_INTERFACE
#define SPI_32MHZ_INTERFACE 0
#endif`

To this:

`// default to 0
#ifndef SPI_32MHZ_INTERFACE
#define SPI_32MHZ_INTERFACE 1
#endif`

If there's a way to implement that in an Arduino sketch, that would be simpler.

dhalbert, 2023-06-09:
I think could just put
```c
#define SPI_32MHZ_INTERFACE 1
```
in your sketch, because it's `#ifndef`'d in the code above, and not defined otherwise for the ItsyBitsy. (For the Feather nRF52840, both `SPI` and `SPI1` are defined.)

amowry, 2023-06-09:
Thanks-- I did try that, both before and after including SPI.h, but it didn't seem to have an effect. Not sure why (?).

hathach, 2023-06-14:
it is probably that 32Mhz SPI need hfclk() which is enabled by USB when detect VBUS, you can try to enable hfclk() to see if that works.

amowry, 2023-06-14:
Thanks! I'd rather not use the hfclk so I'll stay with the previously mentioned fix, but I appreciate the suggestion.

hathach, 2023-06-21:
should be closed now, highspeed 32Mhz needs hfclk to work

closed 2023-06-21 by hathach