adafruit/Adafruit_nRF52_Arduino #787: Building with non-release version leads to platformio assertion error

CSC-Sendance, 2023-08-25:
### Operating System

Windows 11

### IDE version

Platformio 6.1.10

### Board

Feather nrf52840

### BSP version

latest

### Sketch

```
[env]
platform = nordicnrf52
framework = arduino
platform_packages = 
	framework-arduinoadafruitnrf52 @ https://github.com/CSC-Sendance/Adafruit_nRF52_Arduino
```

or 

```
[env]
platform = nordicnrf52
framework = arduino
platform_packages = 
	framework-arduinoadafruitnrf52 @ https://github.com/adafruit/Adafruit_nRF52_Arduino
```

or 

```
[env]
platform = nordicnrf52
framework = arduino
platform_packages = 
	framework-arduinoadafruitnrf52 @ https://github.com/adafruit/Adafruit_nRF52_Arduino#bf5047952f41003603ba55b4530f59b52a81d692
```

### What happened ?

I tried building our project with an unreleased version of the framework using the platformio platform_packages field in 3 variants (github latest, github commit hash, own fork). However, platformio's "Tool Manager" fails due to an assertion error when "Resolving dependencies":
` File "C:\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 72, in from_dependency
    assert isinstance(dependency, dict)`

Executing "Full clean", "Update Dependencies", "PIO Upgrade" etc. in platformio and manually through a fresh install etc. leads to the same result.

Interestingly: Copy&pasting the framework's files to a different repository (not related as fork or similar to this original repository) makes it work for some reason. 

### How to reproduce ?

1. Adapt the platformio.ini with the platform_packages build-options as listed in *Sketch*
2. Try to build
3. Observe the mentioned assertion error (also listed in *Debug log*)

### Debug Log

```
Processing *buildname* (board: adafruit_feather_nrf52840; platform: nordicnrf52; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing git+https://github.com/CSC-Sendance/Adafruit_nRF52_Arduino
git version 2.37.0.windows.1
Cloning into 'C:\.platformio\.cache\tmp\pkg-installing-sian4nd4'...
remote: Enumerating objects: 919, done.
remote: Counting objects: 100% (919/919), done.
remote: Compressing objects: 100% (740/740), done.
remote: Total 919 (delta 253), reused 389 (delta 95), pack-reused 0
Receiving objects: 100% (919/919), 16.09 MiB | 13.91 MiB/s, done.
Resolving deltas: 100% (253/253), done.
Submodule 'libraries/Adafruit_TinyUSB_Arduino' (https://github.com/adafruit/Adafruit_TinyUSB_Arduino.git) registered for path 'libraries/Adafruit_TinyUSB_Arduino'
Submodule 'libraries/Adafruit_nRFCrypto' (https://github.com/adafruit/Adafruit_nRFCrypto.git) registered for path 'libraries/Adafruit_nRFCrypto'
Cloning into 'C:/.platformio/.cache/tmp/pkg-installing-sian4nd4/libraries/Adafruit_TinyUSB_Arduino'...
remote: Enumerating objects: 4790, done.        
remote: Counting objects: 100% (1299/1299), done.        
remote: Compressing objects: 100% (526/526), done.
remote: Total 4790 (delta 928), reused 911 (delta 766), pack-reused 3491        
Receiving objects: 100% (4790/4790), 1.47 MiB | 8.17 MiB/s, done.
Resolving deltas: 100% (2849/2849), done.
Cloning into 'C:/.platformio/.cache/tmp/pkg-installing-sian4nd4/libraries/Adafruit_nRFCrypto'...
remote: Enumerating objects: 327, done.        
remote: Counting objects: 100% (45/45), done.
remote: Compressing objects: 100% (32/32), done.
remote: Total 327 (delta 22), reused 24 (delta 12), pack-reused 282        
Receiving objects: 100% (327/327), 360.87 KiB | 4.75 MiB/s, done.
Resolving deltas: 100% (188/188), done.
Submodule path 'libraries/Adafruit_TinyUSB_Arduino': checked out 'f43338a086764f829afd682dbc5f3abba3f3c881'
Submodule path 'libraries/Adafruit_nRFCrypto': checked out '802d0a6961eae5d1e9bc94a2a570214804e68638'
Tool Manager: framework-arduinoadafruitnrf52@1.0.0+sha.bf50479 has been installed!
Tool Manager: Resolving dependencies...
AssertionError: Traceback (most recent call last):
  File "C:\.platformio\penv\lib\site-packages\platformio\__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\.platformio\penv\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\.platformio\penv\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\.platformio\penv\lib\site-packages\platformio\cli.py", line 85, in invoke
    return super().invoke(ctx)
  File "C:\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\.platformio\penv\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\.platformio\penv\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\.platformio\penv\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\.platformio\penv\lib\site-packages\platformio\run\cli.py", line 145, in cli
    process_env(
  File "C:\.platformio\penv\lib\site-packages\platformio\run\cli.py", line 198, in process_env
    result["succeeded"] = EnvironmentProcessor(
  File "C:\.platformio\penv\lib\site-packages\platformio\run\processor.py", line 81, in process
    install_project_env_dependencies(
  File "C:\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 131, in install_project_env_dependencies
    _install_project_env_platform(project_env, options),
  File "C:\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 148, in _install_project_env_platform
    PlatformPackageManager().install(
  File "C:\.platformio\penv\lib\site-packages\platformio\package\manager\platform.py", line 62, in install
    p.install_required_packages(force=force)
  File "C:\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 76, in install_required_packages
    self.install_package(name, force=force)
  File "C:\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 70, in install_package
    return self.pm.install(spec or self.get_package_spec(name), force=force)
  File "C:\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 47, in install
    pkg = self._install(spec, skip_dependencies=skip_dependencies, force=force)
  File "C:\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 126, in _install
    self.install_dependencies(pkg)
  File "C:\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 139, in install_dependencies
    self.install_dependency(dependency)
  File "C:\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 151, in install_dependency
    dependency_compatibility = PackageCompatibility.from_dependency(dependency)
  File "C:\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 72, in from_dependency
    assert isinstance(dependency, dict)
AssertionError

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html
```

CSC-Sendance, 2023-08-25:
It has to be mentioned trhat I am not 100% certain this is a bug rather than some misconfiguration on my side. We, however, would love to keep a fork of this framework instead of a custom-repository in order to be able to contribute back to the project and also keep an up-to-date version (possibly independent from official Releases)

hathach, 2023-08-25:
try to compile with Arduino IDE or arduino-cli and compare your platformio config with gcc command from arduino. This is probably your mis-configure with platformio. We don't officially support platformio, therefore this is not consiered as a bug. You are welcome to make PR to get this working with pio though. We can merge it if it doesn't require lot of maintanenance from your sidde.

closed since it is not supported. If you are still looking for help, please use discussion instead.

closed 2023-08-25 by hathach