adafruit/Adafruit_nRF52_Arduino #789: Bluefruit.Central.connect Connection Timeout?

macirtr, 2023-09-10:
### Operating System

Windows 10

### IDE version

Ardino 2.2.0

### Board

Adafruit Feather nrf52832

### BSP version

1

### Sketch

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/Central/central_bleuart/central_bleuart.ino

### What happened ?

...
void scan_callback(ble_gap_evt_adv_report_t* report) {
    Bluefruit.Central.connect(report);
}
...
As above, when a device is found that can be connected to, I send the connect command.
If the target device moves away or turns off after sending this command, the connection cannot be established. And the connection hangs. The device cannot be found and connected again.
There should be a connection timeout option here. If the connection does not occur after x milliseconds after the connection is attempted, the operation should be aborted. But the connection hangs.
How do I solve this?

### How to reproduce ?

...

### Debug Log

...

macirtr, 2023-09-18:
https://github.com/nkolban/esp32-snippets/issues/874
I found a thread of programmers having similar problems on Esp32. The semaphore was causing the deadlock. They found a solution by reducing the semaphore time. So I replaced all portMAX_DELAY data in the library with 3000UL, but it didn't work.

hathach, 2023-11-30:
late response, can you post the log with debug level = 2 or 1, to see if there is any error output when the issue occurs.