adafruit/Adafruit_nRF52_Arduino #799: Beacon example fails to compile with BLE_GAP_ADV_TYPE_ADV_NONCONN_IND uncommented

moppymopperson, 2023-11-10:
### Operating System

MacOS

### IDE version

Arduino 2.2.1

### Board

nRF52840 Express

### BSP version

GitHub latest

### Sketch

BlueFruit52Lib/examples/Peripheral/beacon/beacon.ino

### What happened ?

If you uncomment line 82, compilation fails because the macro is undefined.
```
 //Bluefruit.Advertising.setType(BLE_GAP_ADV_TYPE_ADV_NONCONN_IND);
```

I searched the web and found this [definition](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.s132.api.v0.9.0%2Fgroup___b_l_e___g_a_p___a_d_v___t_y_p_e_s.html)

```
#define BLE_GAP_ADV_TYPE_ADV_NONCONN_IND  0x03
```

### How to reproduce ?

1. Uncomment the line above and try to compile the example

moppymopperson, 2023-11-10:
While we're at it, these comments could be more clear about what needs to be changed for Apple platforms.

```java
  /*
   * Apple Beacon specs
   * - Type: Non connectable, undirected
   * - Fixed interval: 100 ms -> fast = slow = 100 ms
   */
  //Bluefruit.Advertising.setType(BLE_GAP_ADV_TYPE_ADV_NONCONN_IND);
  Bluefruit.Advertising.restartOnDisconnect(true);
  Bluefruit.Advertising.setInterval(160, 160);    // in unit of 0.625 ms
```

hathach, 2023-11-29:
should be fixed by #801 , thank you for reporting the issue.

PR #801 (2023-11-29): add more description to beacon example, use non-connectable scannable undirected type
fix #799
closed 2023-11-29 by hathach