adafruit/Adafruit_nRF52_Arduino #800: CircuitPlayground.readCap() not working with Bluefruit.begin()

mikeysklar, 2023-11-15:
### Operating System

MacOS

### IDE version

Arduino 2.2.1

### Board

CircuitPlayground Bluefruit

### BSP version

1.5.0

### Sketch

Simple captouch example works correctly as long as  Bluefruit.begin() is commented out. 

```
// Demo program for testing library and board - flip the switch to turn on/off buzzer

#include <Adafruit_CircuitPlayground.h>
#include <bluefruit.h>

// we light one pixel at a time, this is our counter
uint8_t pixeln = 0;

void setup() {
  Serial.begin(9600);
  Serial.println("Circuit Playground test!");

  CircuitPlayground.begin();
  Bluefruit.begin();
}

void loop() {
  // turn off speaker when not in use
  CircuitPlayground.speaker.enable(false);

  // test Red #13 LED
  CircuitPlayground.redLED(HIGH);
  delay(100);
  CircuitPlayground.redLED(LOW);

  /************* TEST CAPTOUCH */
  Serial.print("Capsense #3: "); Serial.println(CircuitPlayground.readCap(3));
  Serial.print("Capsense #2: "); Serial.println(CircuitPlayground.readCap(2));
  Serial.print("Capsense #0: "); Serial.println(CircuitPlayground.readCap(0));
  Serial.print("Capsense #1: "); Serial.println(CircuitPlayground.readCap(1));
  Serial.print("Capsense #12: "); Serial.println(CircuitPlayground.readCap(12));
  Serial.print("Capsense #6: "); Serial.println(CircuitPlayground.readCap(6));
  Serial.print("Capsense #9: "); Serial.println(CircuitPlayground.readCap(9));
  Serial.print("Capsense #10: "); Serial.println(CircuitPlayground.readCap(10));
  delay(1000);

}
```

[Adafruit Forum Thread](https://forums.adafruit.com/viewtopic.php?t=205852)

### What happened ?

All captouch reads are maxed out when BLE is enabled. Should this work or is BLE always going to prevent captouch from being used?

```
Capsense #10: 65534
Capsense #3: 65534
Capsense #2: 65534
Capsense #0: 65534
Capsense #1: 65534
Capsense #12: 65534
Capsense #6: 65534
Capsense #9: 65534
Capsense #10: 65534
Capsense #3: 65534
Capsense #2: 65534
Capsense #0: 65534
Capsense #1: 
```

### How to reproduce ?

leave Bluefruit.begin() uncommented.

### Debug Log

```
Using library Adafruit Circuit Playground at version 1.11.5 in folder: /Users/sklarm/Documents/Arduino/libraries/Adafruit_Circuit_Playground 
Using library SPI at version 1.0 in folder: /Users/sklarm/Library/Arduino15/packages/adafruit/hardware/nrf52/1.5.0/libraries/SPI 
Using library Wire at version 1.0 in folder: /Users/sklarm/Library/Arduino15/packages/adafruit/hardware/nrf52/1.5.0/libraries/Wire 
Using library Adafruit Bluefruit nRF52 Libraries at version 0.21.0 in folder: /Users/sklarm/Library/Arduino15/packages/adafruit/hardware/nrf52/1.5.0/libraries/Bluefruit52Lib 
Using library Adafruit nRFCrypto at version 0.0.7 in folder: /Users/sklarm/Library/Arduino15/packages/adafruit/hardware/nrf52/1.5.0/libraries/Adafruit_nRFCrypto 
Using library Adafruit TinyUSB Library at version 2.0.1 in folder: /Users/sklarm/Library/Arduino15/packages/adafruit/hardware/nrf52/1.5.0/libraries/Adafruit_TinyUSB_Arduino 
Using library nRF52840 PDM - Adafruit Fork at version 1.0 in folder: /Users/sklarm/Library/Arduino15/packages/adafruit/hardware/nrf52/1.5.0/libraries/PDM 
Using library Adafruit Little File System Libraries at version 0.11.0 in folder: /Users/sklarm/Library/Arduino15/packages/adafruit/hardware/nrf52/1.5.0/libraries/Adafruit_LittleFS 
Using library Adafruit Internal File System on Bluefruit nRF52 at version 0.11.0 in folder: /Users/sklarm/Library/Arduino15/packages/adafruit/hardware/nrf52/1.5.0/libraries/InternalFileSytem 
```

hathach, 2023-11-28:
I made an PR to fix this, but it is the bug of CircuitPlayground library https://github.com/adafruit/Adafruit_CircuitPlayground/pull/70. Would you mind trying out the PR while it is on reviewed and post your result there. 

PS: I am not sure while enabled bluefruit library cause this to occurs, it should always occur though.

mikeysklar, 2023-11-28:
Thank you will try today.

mikeysklar, 2023-11-29:
This #70 PR for Adafruit_CircuitPlayground looks good. I'm seeing clear response to captouch and the values are no longer maxed out at 65k values. 

```
Capsense #9: 23
Capsense #10: 0
Capsense #3: 10
Capsense #2: 831
Capsense #0: 0
Capsense #1: 27
Capsense #12: 15
Capsense #6: 13
Capsense #9: 23
Capsense #10: 0
Capsense #3: 18
Capsense #2: 2466
Capsense #0: 0
Capsense #1: 27
Capsense #12: 46
Capsense #6: 3137
Capsense #9: 23
Capsense #10: 27
Capsense #3: 21
Capsense #2: 5931
Capsense #0: 34
Capsense #1: 19
Capsense #12: 114
Capsense #6: 8343
Capsense #9: 0
Capsense #10: 28
```

hathach, 2023-11-29:
thank you for testing it out.

closed 2023-11-30 by hathach