adafruit/Adafruit_nRF52_Arduino #804: [v1.4.0] Adafruit Feather nRF52840 Express: Cannot link to Serial

zfields, 2023-12-06:
### Operating System

Linux

### IDE version

1.8.19

### Board

Feather nRF52840 Express

### BSP version

1.4.0

### Sketch

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  while (!Serial);
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.println("Hello, World!");
  delay(5000);
}

### What happened ?

Linker error occurs: Cannot link to Serial

I have also tried downgrading to v1.3.0, but that did NOT make a difference.

_Originally opened [issue](https://github.com/adafruit/arduino-board-index/issues/105) on the wrong repository._

### How to reproduce ?

1. Attempt to compile the follow code for the Adafruit Feather nRF52840 Express.

hathach, 2023-12-11:
adding 

```C
#include "Adafruit_TinyUSB.h"
```

will fix this issue.

closed 2023-12-11 by hathach

zfields, 2023-12-12:
Thank you for the workaround :+1:.

Can you please reopen this issue?

The code provided in the example above works on every other platform, so I would consider this a bug. The requirement of including a non-standard library (that you are not aware of) is hard to call a "workable solution".

IMO, this issue should remain open, until the header is included automatically.

hathach, 2023-12-12:
This is not a workaround. It is how it should be on this core.

zfields, 2023-12-12:
I trying to relate to you as a new user that it feels like you are telling me that it should _NOT_ compile for new users who don't understand that `Adafruit_TinyUSB.h` even exist.

My experience with several other Arduino platforms and inexperience with your platform is what is causing this stumbling block for me. I think you my have an unrealized bias based upon your deep experience with the platform.

There is no error message that directs me to the header file you have presented. This serves as unnecessary friction for folks who are trying to adopt your platform.

hathach, 2023-12-13:
It is menteiond in our learn guide, we don't like it either. If you could fix it, please submit a PR.