adafruit/Adafruit_nRF52_Arduino #805: Softwareserial Lib Is Reading Incorrect Values

macirtr, 2023-12-18:
### Operating System

Windows 10

### IDE version

Ardino 2.2.0

### Board

Adafruit Feather nrf52832

### BSP version

1

### Sketch

Adafruit software serial lib

### What happened ?

Softwareserial is reading incorrect value. We use 115200 as the baud rate. Usually the first and last element of the returned byte array are incorrect.
For example, the other device sends 0xBB as the first byte. Serialsoftware sees it as 0x7B.
While the last byte should be 0x7F, the value 0xFE is read.
I investigated the reason for this and found that a timing error occurred in void SoftwareSerial::recv().
delayMicroseconds(_rx_delay_centering);
_rx_delay_centering this value is 2 milliseconds. If I do this for 4 milliseconds, the first byte is read correctly as 0xBB. However, this change disrupts the reading of subsequent bytes.
I tried the ADAFRUIT_TINYUSB library after seeing the error in the library. And it worked flawlessly there. However, rx and tx pins cannot be changed in this library. We should be able to select rx tx pins in the project.
However, I also saw the uart.h code. Pins can be selected in this code. But there are no examples and resources in the folder or on the internet for uart.
Can you help us with this? Do you have a solution suggestion?

### How to reproduce ?

Use 115200 baud rate for swoftware serial lib.
Send a byte array starting with BB from a device using 115200.
You will see that the first byte is read 7B.
delayMicroseconds(_rx_delay_centering); line delayMicroseconds(2); Change it to . You will see that it reads the BB value.

macirtr, 2023-12-21:
After three days, I found the error.

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/SoftwareSerial/SoftwareSerial.cpp
_rx_delay_centering = (bit_delay/2) - 2;

Remove the -2 on line 63 and it works.

hathach, 2023-12-22:
would you mind making an PR for the fix ?