adafruit/Adafruit_nRF52_Arduino #817: BLE UART broken on Android 14+ OneUI 6+

rsauchyn, 2024-03-21:
### Operating System

Others

### IDE version

1.8.15

### Board

Custom nRF52840 data logger, Bluefruit firmware

### BSP version

github latest

### Sketch

while (dataFile.available()) {
  bleuart.write(dataFile.read());
}

// Further source available upon request

### What happened ?

Is anyone else finding that OneUI 6 breaks BLE UART? Only the first packet comes through when writing >328B.

### How to reproduce ?

OneUI 6 (avail on Android 14+)

closed 2024-04-04 by rsauchyn

rsauchyn, 2024-07-25:
@rgatej We resolved it by writing at most MTU bytes at a time, and flushing TXD after each write. Doesn't appear to affect performance. It's clearly something about the underlying packet handling (also seems to affect the Pixel 8). Curious if this resolves the issue for you.

```
while (dataFile.available()) {
  bufCounter = dataFile.read(tmp, 240); // must be <= MTU
  if (bufCounter > 0) { // 0=EOF, -1=ERR
    bleuart.write(tmp, bufCounter);
    bleuart.flushTXD(); // fix OneUI 6+
  }
}
```

rgatej, 2024-07-26:
@rsauchyn The flushTXD() call did it for me (I was already writing 244 bytes at a time). I am only seeing this on Android 14+ (on Motorola and Oppo, I haven't tried with OneUI).

Thanks a lot for the quick response!