adafruit/Adafruit_nRF52_Arduino #824: With v.1.6.1 there is a delay while connecting to USB

amowry, 2024-06-01:
### Operating System

Windows 11

### IDE version

2.3.2

### Board

ItsyBitsy NRF52840 Express

### BSP version

1.6.1

### Sketch

Custom NRF52840 board: https://github.com/amowry/WARBL2

### What happened ?

After updating from 1.6.0 to 1.6.1, my custom USB MIDI board using the ItsyBitsy NRF52840 Express takes several seconds to establish a USB connection. Do have any suggestions for what might cause this?

### How to reproduce ?

Create USB MIDI device using the ItsyBitsy NRF52840 Express package

amowry, 2024-06-03:
The problem seems to be with TinyUSB v 3.1.3 or higher, so I've logged a bug there.

amowry, 2024-07-08:
This will be fixed with this TinyUSB pull request:

https://github.com/adafruit/Adafruit_TinyUSB_Arduino/pull/437

I'm not sure how many users this affects, but possibly roll back the TinyUSB version associated with this BSP until it is fixed? Thanks!

SlayerPower, 2024-10-02:
OMG COLO LO HACES